HaythamB/TAMV #20: bad detection of big nozzle sizes

well-engineered, 2021-03-29:
The nozzle orifice of a big volcano nozzle seems be hard to detect reliably.

HaythamB, 2021-03-29:
Currently looking into better methods for nozzle detection so we can handle these cases better. Just to reiterate what we had learned: large nozzle diameters when in perfect focus of the microscope, exhibit some sort of ghost artifacts at the center of the nozzle under the "Xray" view. This seems like it may be reflections from the brass nozzle sides bouncing back against the microscope sensor and triggering these odd shapes in the center of an otherwise solid black nozzle.

HaythamB, 2021-10-10:
Just an update: replicated the same issue on a 0.8 nozzle. Should be fixed very soon..

HaythamB, 2022-09-09:
dev_alpha branch with updated nozzle detection parameters and new bTriangle approach should fix this.

closed 2022-09-09 by HaythamB