HaythamB/TAMV #28: M551 Conflict

HebigT, 2021-06-14:
Using TAMV with great success but noticed this:

I had M551 and a password in my config ( i'm on a shared network).

This prevented TAMV from connecting to the duet. (Duet 2 RRF 3.2.2)

https://reprap.org/wiki/G-code#M551:_Set_Password

HaythamB, 2022-09-11:
Its quite late from when this was submitted, but its working with passwords now. Defaults to "reprap" as per Duet's configurations, and you can use your own password. 

Tested on my Duet 2 board, and should work for Duet 3 boards in standalone mode.

closed 2022-09-11 by HaythamB

HebigT, 2022-09-13:
> Its quite late from when this was submitted, but its working with passwords now. Defaults to "reprap" as per Duet's configurations, and you can use your own password.
> 
> Tested on my Duet 2 board, and should work for Duet 3 boards in standalone mode and in SBC mode

Awesome, thank you for the update!

commit c81f0c37 (2022-09-11): closed #28

added password support for Duet 2 boards and Duet 3 boards using rr_status interface (standalone mode)