HaythamB/TAMV #46: Hard-to-understand camera-not-connected message

zruncho3d, 2022-09-13:
When starting the TAMV GUI, with no webcam connected, this is the message one sees:

```bash
...
WARNING  : This is an alpha release. Always use only when standing next to your machine and ready to hit EMERGENCY STOP.
INFO     : Launching application.. 
INFO     :   .. reading drivers..
INFO     :   .. creating new settings.json..
INFO     :   .. starting video feed.. 
Traceback (most recent call last):
  File "TAMV_GUI.py", line 3659, in <module>
    a = App()
  File "TAMV_GUI.py", line 2469, in __init__
    self.startVideo()
  File "TAMV_GUI.py", line 3580, in startVideo
    self.video_thread = CalibrateNozzles(parentTh=self,numTools=0, cycles=1, align=False)
  File "TAMV_GUI.py", line 928, in __init__
    self.change_pixmap_signal.emit(local_img)
TypeError: CalibrateNozzles.change_pixmap_signal[numpy.ndarray].emit(): argument 1 has unexpected type 'NoneType'
```

Would be nice to catch the error and suggest something to the user.

HaythamB, 2022-09-13:
yes I'm working on this, since this error comes from openCV and I have to figure out how to catch its exceptions.

HaythamB, 2022-09-13:
Won't be able to fix this any time soon, as it will require a lot of exception handling all over the place to allow openCV to throw exceptions (as in every time we fetch a frame). Its do-able, but not worth the effort in the short term.

My guess as to this error occuring is most likely an issue opening the specified camera feed. 

Just to address this: once the 2.1 release camera issues get addressed to allow multiple sources, a lot of the error handling will be taken care of in a single location, and the likelihood of this exception being raised unhandled will be almost zero.

HaythamB, 2022-12-04:
So TAMV in its current release will exit with a critical error if there's no webcam connected. Its part of it detecting camera errors during runs, and its a bit complicated to untangle things to allow you to run it without any camera connected. If its any help, when I'm doing dev work that doesn't need nozzle detection, I simply redirect it to read from my webcam, as that shows up as a normal camera feed and allows me to work on other aspects of TAMV.

tldr: connect any camera source (even a virtual OBS source if you have to) if you want to just run the program to poke about.

closed 2022-12-04 by HaythamB